HelTecAutomation/Heltec_ESP32 #1: No member named 'LoRa' - LoRaSender example

QuentinFombaron, 2019-02-04:
Hi, 

I tried to run LoRaSender exemple from "`Examples/Heltec ESP32 Dev-Boards/LoRa`" and it shows me an error: 

```
/Users/Quentin/Documents/Arduino/libraries/Heltec_ESP32_Dev-Boards/examples/LoRa/LoRaSender/LoRaSender.ino: In function 'void loop()':
LoRaSender:28: error: 'class Heltec_ESP32' has no member named 'LoRa'
   Heltec.LoRa.beginPacket();
          ^
LoRaSender:29: error: 'class Heltec_ESP32' has no member named 'LoRa'
   Heltec.LoRa.print("hello ");
          ^
LoRaSender:30: error: 'class Heltec_ESP32' has no member named 'LoRa'
   Heltec.LoRa.print(counter);
          ^
LoRaSender:31: error: 'class Heltec_ESP32' has no member named 'LoRa'
   Heltec.LoRa.endPacket();
          ^
exit status 1
'class Heltec_ESP32' has no member named 'LoRa'
```

Is something missing on `heltec.h`?

Best regards,

Quentin

Marcelovds, 2019-02-04:
Hi Quentin, I´m facing the same issue.    
I´m reading the source files and trying to figure out what is going on.  
I guess it make take some time go get support from Heltec as they are in New Year celebration.  
gds, Marcelo

Marcelovds, 2019-02-05:
Quentin,  try this:

Include the lib LoRa.h right after heltec.h:
#include "lora/LoRa.h"

then, remove all "Heltec." from all   "Heltec.Lora...........".
Example:
change from: 

Heltec.LoRa.beginPacket();

change to:
LoRa.beginPacket();

Let me know your findings.  
best regards from Brazil!

QuentinFombaron, 2019-02-05:
I tried both of your solutions, it compiles but I got this error on the serial monitor : 
```
Serial initial done
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d11ef  PS      : 0x00060130  A0      : 0x800d0c9d  A1      : 0x3ffb1f30  
A2      : 0x00000000  A3      : 0x3f400038  A4      : 0x0800001c  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800d1d60  A9      : 0x3ffb1f10  
A10     : 0x00000015  A11     : 0x3f400038  A12     : 0x3ffbdca0  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa  

Backtrace: 0x400d11ef:0x3ffb1f30 0x400d0c9a:0x3ffb1f50 0x400d0ba9:0x3ffb1f90 0x400d2023:0x3ffb1fb0 0x40086d41:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698
```

cortinas, 2019-02-05:
I had that also.... I changed the Tools-> 'flash freq' to 40 and it fixed mine...
Honestly not sure why

QuentinFombaron, 2019-02-05:
Thank you @cortinas but it seems to not working... 
[image]

QuentinFombaron, 2019-02-05:
So...

- I am using the Heltec WiFi LoRa 32 (V1) card.

- It is my code of modified LoRaSender example (`Examples/Heltec ESP32 Dev-Boards/LoRa`):
```
/* 
  Basic test program, send date at the BAND you seted.
  
  by Aaron.Lee from HelTec AutoMation, ChengDu, China
  成都惠利特自动化科技有限公司
  www.heltec.cn
  
  this project also realess in GitHub:
  https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series
*/
#define WIFI_LoRa_32

#include "heltec.h"
#define BAND 915E6 //you can set band here directly,e.g. 868E6,915E6

int counter = 0;

void setup() {
  //WIFI Kit series V1 not support Vext control
  Heltec.begin(false /*DisplayEnable Enable*/, true /*Heltec.LoRa Disable*/, true /*Serial Enable*/, true /*PABOOST Enable*/, BAND /*long BAND*/);
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);
  // send packet
  Heltec.LoRa.beginPacket();
  Heltec.LoRa.print("hello ");
  Heltec.LoRa.print(counter);
  Heltec.LoRa.endPacket();
  
  counter++;
  digitalWrite(25, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(25, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

- I comment `VextOn()` and `VextOff()` functions on heltec.h (l.30-31 )and heltec.cpp (l.24 and l.89-99).

- It compiles but when I upload the code, a restart loop is shown:
```
 Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
Serial initial done
Sending packet: 0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1d44  PS      : 0x00060f30  A0      : 0x800d1d60  A1      : 0x3ffb1f50  
A2      : 0x3ffbfcb4  A3      : 0x3f400031  A4      : 0x00000000  A5      : 0x3ffbff74  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1f40  
A10     : 0x00000006  A11     : 0x3f400037  A12     : 0x00000006  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x400d1d44:0x3ffb1f50 0x400d1d5d:0x3ffb1f70 0x400d0bdb:0x3ffb1f90 0x400d2035:0x3ffb1fb0 0x40086d41:0x3ffb1fd0
```

- I tried to flash the card with  `esptool.py erase_flash`, I also change my USB cable.

I start to believe I am doing something wrong... :(

QuentinFombaron, 2019-02-05:
Yes you're right it is working when I comment the `Heltec.LoRa.print()` lines but it is the principal command of the example... I didn't have the idea to use the ESP Exception Decoder :) 
I am trying to resolve this issue cause it also appear when I am trying to get GPS location with an external chip.

cortinas, 2019-02-05:
Just checking the LoRa library and there is no print() method !
I guess it should be Heltec.LoRa.write(...) but the content is typed as a pointer to uint_8

size_t LoRaClass::write(const uint8_t *buffer, size_t size)

cortinas, 2019-02-05:
PS. I also use these LoRa boards -> https://lowpowerlab.com/shop/product/99
Work well with radiohead lib and better tech support ;-)

dangerousgirl, 2019-02-06:
I have similar issue with V2.  I tried everything here, no joy.  Following this thread with hope.  :)

Heltec-Aaron-Lee, 2019-02-16:
@QuentinFombaron @alongcameastranger @cortinas 
You must using [this develop environment](https://github.com/espressif/arduino-esp32) provide by espressif? 

In this new library, we used many selective compilation for various boards, the espressif project still didn't inlcude that. I had submission a push request to that project few days ago, but still not allowed.
[image]
[image]

To solve this problem temporarily, I recommend you use this develop environment: [https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series
)

Here is the install manual: [http://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino](http://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino)

Sorry for the late reply, we had a long Spring Festival holiday, it's a national holiday.

closed 2019-02-17 by Heltec-Aaron-Lee