HelTecAutomation/Heltec_ESP32 #2: error: 'Vext' was not declared in this scope

elennaro, 2019-02-13:
Board: HTIT-WB32
PlatformIO: version 3.6.4
CLion: 2019.1 EAP
OS: fedora 29

Steps to reproduce:

platformio.ini
```
[env:heltec_wifi_kit_32]
platform = espressif32
board = heltec_wifi_kit_32
framework = arduino

lib_deps =
    Heltec ESP32 Dev-Boards
```

src/main.cpp

```
#include "heltec.h"

void setup() {
    Heltec.display->init();
    Heltec.display->flipScreenVertically();
    Heltec.display->setFont(ArialMT_Plain_10);
    Heltec.display->drawString(0, 0, "Hello dude!");
}

void loop() {
}
```
on build I've got this error:

```
Indexing .pioenvs/heltec_wifi_kit_32/lib3a1/libWire.a
Compiling .pioenvs/heltec_wifi_kit_32/FrameworkArduino/IPAddress.cpp.o
.piolibdeps/Heltec ESP32 Dev-Boards_ID6051/src/heltec.cpp: In member function 'void Heltec_ESP32::VextON()':
.piolibdeps/Heltec ESP32 Dev-Boards_ID6051/src/heltec.cpp:91:10: error: 'Vext' was not declared in this scope
pinMode(Vext,OUTPUT);
^
.piolibdeps/Heltec ESP32 Dev-Boards_ID6051/src/heltec.cpp: In member function 'void Heltec_ESP32::VextOFF()':
.piolibdeps/Heltec ESP32 Dev-Boards_ID6051/src/heltec.cpp:97:10: error: 'Vext' was not declared in this scope
pinMode(Vext,OUTPUT);
^
Compiling .pioenvs/heltec_wifi_kit_32/FrameworkArduino/IPv6Address.cpp.o
*** [.pioenvs/heltec_wifi_kit_32/libafc/Heltec ESP32 Dev-Boards_ID6051/heltec.cpp.o] Error 1
========================== [ERROR] Took 1.93 seconds ==========================
gmake[3]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/build.make:57: CMakeFiles/PLATFORMIO_BUILD] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:221: CMakeFiles/PLATFORMIO_BUILD.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:233: CMakeFiles/PLATFORMIO_BUILD.dir/rule] Error 2
gmake: *** [Makefile:170: PLATFORMIO_BUILD] Error 2
```

Goliath86, 2019-02-15:
Same for me. I've added this line of code inside the heltec.h file and all is compiling now:
`#define Vext 21`

elennaro, 2019-02-15:
@Goliath86 thanks a ton for the hint!
and why 21?

Goliath86, 2019-02-15:
> @Goliath86 thanks a ton for the hint!

Because 21 is the digital pin of the board that enable/disable the Vext. Only the **Heltec LoRa 32 V2** support the Vext functions of the library

Heltec-Aaron-Lee, 2019-02-16:
@elennaro Hi there, @Goliath86 had given a right solution.
Because the pinMap from espressif didn't include our V2 board, use the develop enviroment provided by us will not have such an error.
[https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series)
Here is install manual: [http://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino](http://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino)

closed 2019-02-16 by Heltec-Aaron-Lee

elennaro, 2019-02-16:
Thank you @Heltec-Aaron-Lee , I jus didn't get how to install https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series on platformio.