HelTecAutomation/Heltec_ESP32 #4: LoRa OLED code reboots again and again

madhujiot, 2019-03-03:
Trying to run on my ttgo esp32 Lora board, the OLED LoRa receiver code results in rebooting again and again, please provide a fix. thanks.

Heltec-Aaron-Lee, 2019-03-04:
Power supply problem, make sure the power supply ≥5V, 1A

madhujiot, 2019-03-07:
No Aaron, this specific code am having problem... Other similar code am
able to run...

Heltec-Aaron-Lee, 2019-03-12:
@madhujiot only OLED relative code can't run?

Marcelovds, 2019-03-12:
@madhujiot , isn´t it because you are using TTGO board?   I have the heltec v2 and it´s working here.  I remember when I tried TTGO that there were some pins that must be changed.

madhujiot, 2019-03-18:
It worked eventually but not this code, will share you the one worked
Thanks

stefkpl, 2019-03-27:
Try to add under Heltec.begin( .....

```
  Heltec.display->init();
  Heltec.display->flipScreenVertically();  
  Heltec.display->setFont(ArialMT_Plain_10);
```

I send a merge request today.