HelTecAutomation/Heltec_ESP32 #5: OLED code interferes with ADS1115

haganwalker, 2019-03-07:
Hi there - I love the Heltec ESP32 board with LoRa. Thank you for manufacturing. I am having one problem. I am trying to use an ADS1115 over i2c and if the OLED is enabled, the ADS1115 always reports 65535 (the max number for an int) or -1. If the OLED is disabled, the ADS1115 functions as intended. It seems like something in the OLED .h or .cpp is interfering with other items on the i2c bus, but I'm not sure exactly what. I have made sure that the i2c items are on different addresses. Here is some quick sample code to reproduce the issue:

`/*
 * HelTec Automation(TM) ESP32 Series Dev boards OLED draw Simple Function test code
 *
 * - Some OLED draw Simple Function function test;
 *
 * by LXYZN from HelTec AutoMation, ChengDu, China
 * 
 * www.heltec.cn
 *
 * this project also realess in GitHub:
 * https://github.com/HelTecAutomation/Heltec_ESP32
*/

// This example just provide basic function test;
// For more informations, please vist www.heltec.cn or mail to support@heltec.cn

#include "Arduino.h"
#include "heltec.h"
#include "images.h"
#include <Adafruit_ADS1015.h>

Adafruit_ADS1115 ads(0x4B);

#define DEMO_DURATION 3000
typedef void (*Demo)(void);

int demoMode = 0;
int counter = 1;

void setup() {
  Heltec.begin(true /*DisplayEnable Enable*/, false /*LoRa Disable*/, true /*Serial Enable*/);
  ads.setGain(GAIN_FOUR);       // 4x gain   +/- 1.024V  1 bit = 0.03125mV
  ads.begin();

  Heltec.display->flipScreenVertically();
  Heltec.display->setFont(ArialMT_Plain_10);

}

void loop() {
  // clear the display
  Heltec.display->clear();
  
  int16_t adc0;
  adc0 = ads.readADC_SingleEnded(0);
  int16_t cm = adc0 / 100;
  int16_t inches = cm * 0.3937;
  //adc1 = ads.readADC_SingleEnded(1);
  //adc2 = ads.readADC_SingleEnded(2);
  //adc3 = ads.readADC_SingleEnded(3);
  Heltec.display->drawString(0, 20, "Dist: ");
  Heltec.display->drawString(60, 20, String(adc0));
  Heltec.display->drawString(110, 20, "in");
  Heltec.display->display();
  Serial.print("AIN0: "); Serial.println(adc0);
  delay(1000);
}`

Heltec-Aaron-Lee, 2019-03-21:
Hi @haganwalker , I think the problem is caused because you did make clear or stop I2C bus at the proper time.

Anyway, you can use another I2C bus. there was three I2C bus in WIFI LoRa 32

cdc31, 2019-06-05:
Hi, did you manage to solve this issue? If yes, could you please describe how?

haganwalker, 2019-06-05:
Unfortunately, I did not. Something with the screen code is causing this overflow. A workaround: If you disable the “true” setting for using the display on the heltec enable line, you don’t get the overflow, but you also don’t get to use the display. It does work with serial, etc though.

Heltec-Aaron-Lee, 2019-06-05:
@cdc31 @haganwalker 
You can try to use I2C1, here is a basic example code: [https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/ESP32/I2C1_Scanner/I2C1_Scanner.ino](https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/ESP32/I2C1_Scanner/I2C1_Scanner.ino)

a2ruan, 2020-11-27:
I just wanted to say to use pins 15 and 2.  You can piggy back off the OLED I2C channels and it should work for ESP32 LORA V2