HelTecAutomation/Heltec_ESP32 #6: FactoryTest for Lora 32 v2 doesn't work.

tavdog, 2019-03-12:
I have two Lora V2 boards.  They do work with some sketches but the factory ping pong test does not work.  The factory test DID work out of the box, but once flashed with the example code here, both boards fail to receive any packets.  They send the packets but do not receive.

Heltec-Aaron-Lee, 2019-03-21:
Hi @tavdog 
did you install the newest [Heltec ESP32 develop environment](https://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino)?

[image]
Also you need config here to the RF band you need. I think 868E6 is fine for you.

stefkpl, 2019-03-27:
Try to add under Heltec.begin( .....

```
  Heltec.display->init();
  Heltec.display->flipScreenVertically();  
  Heltec.display->setFont(ArialMT_Plain_10);
```
I send a merge request today.

tavdog, 2019-03-27:
Why would adding display commands fix the RF sending problems ?  I am still having problems with the heltec boards.  They won't transmit after a receive.  If software reset then transmitting works again.

stefkpl, 2019-03-28:
Sorry, it's mistake, my reply was for the other post :(
I use Arduino IDE with the serial monitor. In this case, you can see errors from your board.

Heltec-Aaron-Lee, 2019-03-29:
@tavdog We released V1.0.5 library with LoRa bug fix, did you try this new version?

tavdog, 2019-03-29:
I have tried the new version. The FactoryTest example still does NOT work.  The OLED Lora Receive / Transmit examples DO work.

Heltec-Aaron-Lee, 2019-03-29:
@tavdog Oh, how come... I will test it now

tavdog, 2019-03-29:
The Lora part of the FactoryTestLora is working but the received packets are only showing in the serial output, not on the OLED screen.

solotaker, 2019-03-29:
> The Lora part of the FactoryTestLora is working but the received packets are only showing in the serial output, not on the OLED screen.

Sorry, I fixed this problem, and now the example works fine. The reason for the problem is because there were some errors in the previous code submission.

closed 2019-03-29 by solotaker