HelTecAutomation/Heltec_ESP32 #9: [U8x8lib] oled problem

DaveCalaway, 2019-03-28:
Hi everyone, 
i want to use the monitor on the board to display some informations.

With this simple code, the esp32 crash:
```
#include <U8x8lib.h>

U8X8_SSD1306_128X64_NONAME_HW_I2C display(/*rst*/ 16, /*scl*/ 15, /*sda*/ 4);

void setup() {
  display.begin();
  delay(500);
}

void loop() {
  display.drawString(0, 0, "Starting");
  delay(1000);
}
```

This is the error:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0ff8  PS      : 0x00060d30  A0      : 0x800d105c  A1      : 0x3ffb1f40  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f400020  
A6      : 0x0000fffe  A7      : 0x3ffbfcc0  A8      : 0x8008adf6  A9      : 0x3ffb1f20  
A10     : 0x00000001  A11     : 0x00000410  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060f20  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000002  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d0ff8:0x3ffb1f40 0x400d1059:0x3ffb1f70 0x400d0d4f:0x3ffb1f90 0x400d2755:0x3ffb1fb0 0x4008919d:0x3ffb1fd0
```

I think that the pinOut is correct to the display, but i don't understand the error message.

stefkpl, 2019-03-28:
Hi, you must install this : https://github.com/me-no-dev/EspExceptionDecoder
After compilation, run and copy/past in exception decoder windows. Very easy to found errors ;-)
Stef

DaveCalaway, 2019-03-29:
Solution,  i must add the setFont:

```
....
u8x8.begin();
u8x8.setFont(u8x8_font_chroma48medium8_r);
.....
```