HelTecAutomation/Heltec_ESP32 #15: Way too low RSSI

jayanta525, 2019-06-01:
Two Heltec ESP32 LoRa boards, 

- RSSI -19 when both antennae are placed touching each other parallel,
- 1cm distance placed parallel RSSI -31,
- 15cm distance placed parallel RSSI -51.

Is this problem with the board or do I need a different library? 

Things I tried: 

- Different library
- Different band
- Checked antenna

Heltec-Aaron-Lee, 2019-07-18:
What band did you use?
Is the working band suitable for your hardware?
Did you assemble the antenna?

Barmatuhin, 2019-09-08:
The same issue here. Very poor RSSI. I've ordered Heltec Wireless Stick Lite for 868MHz with antenna in box. 
Try to change SF, bandwith, TX power... looks like nothing take changes

Heltec-Aaron-Lee, 2019-09-09:
[https://docs.heltec.cn/#/en/user_manual/lora_maximum_communication_distance_test](https://docs.heltec.cn/#/en/user_manual/lora_maximum_communication_distance_test)

Can you please refer to this document? Maybe it help you.

Barmatuhin, 2019-09-09:
Well... i can say that in mostly all done as described in this document. 
Framework and Library added as described in documentation for my Wireless Stick Lite  on heltec.org site.
I take examples for LowPower transmitter side and make few modifications for my project, but main things for TxPower, BAND frequency are still untouched.
Antenas placed side by side in 1 cm between them. BAND seletect to 868E6 on both of them.
The best values of RSSI i saw is -26dbi, and when distance between two antenas is  10cm the RSSI falls to -109 ... -112
I have two doubts about this situation:
1. seller send me a chinees version of Wireless Stick Lite. Small plastick transparent box have a label "Wireless_Stick_Lite_470"  ?! but inside the box i have antena for 868-915Mhz...
2. My scatch (defined heltec.h library) don't accept any settings about TxPower, speadfactor, bandwith... e.t.c, because when i change setTxPower from 14 to 20 nothing changes with RSSI. SF also must give good results, but when i set maximum SF 12 also no significant changes in RSSI (all changes in measurment error range)

Barmatuhin, 2019-09-14:
Hm, i take another board Heltec Wifi LoRa32 as sender (it's marked on pcb 868-915Mhz) and TTGO board as receiver and use heltec examples for sender and receiver side. On both of them set freq 868E6 and on sender side TxPower to 20.
Get the same poor RSSI about -60db when antenas are side to side....
Need help....

mauroscianca98, 2019-11-18:
same thing here with two heltec v2... awful LoRa rssi

PatricioCoronado, 2019-12-28:
Same problem with Heltec WifiLoRa V2. Side to side and RSSI=-104

Heltec-Aaron-Lee, 2020-03-13:
Had updated the code about RSSI calculation part

closed 2020-07-17 by jayanta525