HelTecAutomation/Heltec_ESP32 #16: WiFi_LoRa_32 Factory Test crashes with Guru Meditation number

imakershop, 2019-06-25:


imakershop, 2019-06-25:
After running for several minutes the code crashes with this output in the serial monitor window:

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401433a0: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegaGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered () 
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1⸮@⸮⸮?d (Double exception)
Guru Meditation @⸮⸮⸮?⸮⸮⸮? 1˚⸮x⸮⸮?⸮⸮⸮?⸮⸮? %⸮⸮?  H⸮⸮?Guru Meditation @⸮⸮⸮?⸮⸮⸮? 1˚⸮x⸮⸮?⸮⸮⸮?⸮⸮? Guru Meditation @⸮⸮⸮?⸮⸮⸮? 1˚⸮x⸮⸮?⸮⸮⸮?⸮⸮? Guru Meditation @⸮⸮⸮?⸮⸮⸮? 1˚⸮x⸮⸮?⸮⸮⸮?⸮⸮? Guru Meditation @⸮⸮⸮?⸮⸮⸮? 1˚⸮x⸮⸮?⸮⸮⸮?⸮⸮? Guru Meditation @⸮⸮⸮?⸮⸮⸮? 1˚⸮x⸮⸮?⸮⸮⸮?⸮⸮? Guru Meditation @⸮⸮⸮?⸮⸮⸮? 1˚⸮x⸮⸮?⸮⸮⸮?⸮⸮? Serial initial done
you can see OLED printed OLED initial done!

I have 2 boards and have only observed it on one of them so far. Any clues?

imakershop, 2019-06-25:
The output at the time of the crash: 
<snip>....
hello 1703
hello 1708
hello 1709
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401433a0: bad00bad bad00bad bad00bad
...

Heltec-Aaron-Lee, 2019-07-18:
Hi, there

This document may help you:
[https://docs.heltec.cn/#/en/faq/heltec_esp_series_common_problem_summary?id=system-reset-all-the-time](https://docs.heltec.cn/#/en/faq/heltec_esp_series_common_problem_summary?id=system-reset-all-the-time)

closed 2020-05-30 by imakershop