HelTecAutomation/Heltec_ESP32 #29: Why fragement the arduino-LoRa library?

morganrallen, 2020-01-23:
Hi, I'm Morgan, one of the maintainers of the `arduino-LoRa` library. It's just come to my attention you have copied the driver code into your code instead of linking to the original through the Arduino Library Manager. I'm wondering why? This is bad for both of our communities as it can lead to confusion, duplicate work and fragmentation. I can already see extra manual work done in your code that was already done in the original library, #24 for example was done almost a year before.

It would be great if you could update your documentation to install the `arduino-LoRa` library so everyone can get the best of LoRa on the ESP32.

p.s. the original support I added for ESP32 was done using HelTec dev boards.

Mihaylov93, 2021-11-23:
I was wondering the same, when I found out some things like `_onTxDone` are not implemented in the current version of this repo.

morganrallen, 2021-11-25:
They also stripped the license info from the headers, which is downright illegal.