HelTecAutomation/Heltec_ESP32 #31: Monitor VBAT on ESP32

quietbiker, 2020-02-25:
Using the example code, VBAT is reported as 9214mV

This is with the 1000mAh lion battery attached (usb removed when battery connected).

My board is the ESP32 WIFI KIT 32

mattcalavera, 2020-02-28:
Me too has the same problem. I'm on a Wifi Lora32 (V2) and the battery reading with the provided sketch (battery_power) provide wrong data. Any suggestion?

mattcalavera, 2020-02-28:
And why the divider coefficient is XS = 0.00225 when the divider in the scheme is 110KOhm/220KOhm??

rliegmann, 2020-04-09:
Hey, same Problem here.
Has someone found a solution?
Thanks

henrytheshed, 2020-05-04:
Hi
I have a number of comments.
1) The schematic still shows the battery monitoring pin as 13 where it is now 37. The schematic and pinout needs updating.
2) The 9volt value is equivalent to 4095 which is invalid in some way.  Irt only started to be displayed when I switched on Wifi. Even though Im using ADC 1 I assumed some sort of error.  I adjusted the clock div parameter from 255 to 1 and the voltage value returned to a reasonable value!
3) I have no idea why that worked.  I would quite like a qualified answer!