HelTecAutomation/Heltec_ESP32 #40: Outdated links in repo !!

vincegre, 2020-06-09:
Hi

Is that repo still maintained ?? Numerous links are just 404 !!

`http://docs.heltec.cn/#/en/user_manual/how_to_install_esp32_Arduino?id=_1-execute-a-example-likes-factorytestino`

Detailed tuto for an example that would be very useful: 404 !

`https://heltec.org/summary-of-common-problems-in-wifi-kit-series-continuous-update/`

Summary of common problems: 404 !

And Chinese version is same :(

Code is poorly written (reference at AzurIoT library in module that is no more there !) and no way to compile it unless having followed the only tutorial that I could find :-1: 

Someone succeeds to use these Heltec products or it's just a nightmare ?

Vincèn

Heltec-Aaron-Lee, 2020-06-11:
Hi, @vincegre 

Very sorry about the problem you have encountered, the quick for ESP32 LoRa had moved to here: [https://heltec-automation-docs.readthedocs.io/en/latest/esp32/quick_start.html](https://heltec-automation-docs.readthedocs.io/en/latest/esp32/quick_start.html)

BTW. Here is the new document page: [https://heltec-automation-docs.readthedocs.io/en/latest/index.html](https://heltec-automation-docs.readthedocs.io/en/latest/index.html)

Heltec-Aaron-Lee, 2020-06-11:
Just updated the README.md file.

Thank you for your feedback. Your feedback makes things become better.

closed 2020-06-27 by vincegre