HelTecAutomation/Heltec_ESP32 #54: Compile Error: SSD1306SimpleDemo.ino

o-bri, 2020-10-09:
I tried to compile the file: SSD1306SimpleDemo.ino 
in Arduino IDE, but it shows a compile error:

`Arduino: 1.8.13 (Linux), Board: "Heltec WiFi Kit 32, Disabled, Default, 240MHz (WiFi/BT), QIO, 40MHz, 4MB (32Mb), 921600, None"

In function 'void drawImageDemo()':
oled_ESP32:118:37: error: 'WiFi_Logo_width' was not declared in this scope
     Heltec.display->drawXbm(34, 14, WiFi_Logo_width, WiFi_Logo_height, WiFi_Logo_bits);
                                     ^
oled_ESP32:118:54: error: 'WiFi_Logo_height' was not declared in this scope
     Heltec.display->drawXbm(34, 14, WiFi_Logo_width, WiFi_Logo_height, WiFi_Logo_bits);
                                                      ^
oled_ESP32:118:72: error: 'WiFi_Logo_bits' was not declared in this scope
     Heltec.display->drawXbm(34, 14, WiFi_Logo_width, WiFi_Logo_height, WiFi_Logo_bits);
                                                                        ^
exit status 1
'WiFi_Logo_width' was not declared in this scope
`

edjo82, 2021-09-07:
are you sure you have the correct board installed on Arduino for your Heltec?

o-bri, 2021-09-07:
TBH I can not remember how I solved the problem, as it is too long ago.
So I will close the issue.

closed 2021-09-07 by o-bri