HelTecAutomation/Heltec_ESP32 #61: LoRa receiver (running example) drops OLED and Serial print

frasermarlow, 2021-02-02:
I am working with two Heltec LoRa 32 (v2) boards 
Programming on Arduino IDE 1.8.13
heltec.h (ESP32 Dev-Boards library) is version 1.1.0
Running the example found under `heltec ESP 32 Dev boards > LoRa>  LoraReceiver & LoRaSender`  (without modification).
Powered via USB from a computer (also tried USB powered from a wall socket and off a 3.7V 650 mAh battery)

The issue I am encountering is that after a seemingly random period of time the `receiver` board no longer updates the OLED display or the serial port comms.  
I added an LED blinker to the code, and the loop() is still executing as the LED is blinking...

I have tried swapping the boards around several times (Sender <> Receiver) and it's only the receiver that freezes. 

A hard reboot of the board works fine and it catches up with the sender almost immediately, but then given enough time will just freeze up again.  

I have searched the other posts but can't find something similar reported.  Is this a known issue?  Any suggestions on how to resolve this?  Thanks.

Cenda608, 2021-02-25:
I have same problem from beginning, on this Heltec LoRa 32v2 and nodeMCU(atmega328P+Ra-02). Heltec.h is few libraries together. Now I am using standalone libraries and it works great.
For LoRa you can try Sandeepmistry (github.com/sandeepmistry/arduino-LoRa) or Radiolib (github.com/jgromes/RadioLib)
Radiolib is better for me. Small setting example:

#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>   // for OLED
#include <WiFi.h>   // for wifi
#include <RadioLib.h>   // for LoRa

// OLED
#define OLED_SDA 4
#define OLED_SCL 15 
#define OLED_RST 16
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
**Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RST);**

// LORA
// SX1278 has the following connections:
// NSS pin:   18
// DIO0 pin:  26
// RESET pin: 14
// DIO1 pin:  35
**SX1278 radio = new Module(18, 26, 14, 35);**   // i have SX1278 433MHz

Excuse my english and I am new here.

krysatpat, 2021-03-22:
I have the same issue, I had to detect the last time it receive and reinitialize lora after some time.

fjohannes, 2021-08-29:
Having the same issue! :/
Will try with standalone libraries now...

ChangeD20, 2021-08-30:
Same issue. 

It seems it got much better after adding a yield(); after the line with LoRa.parsePacket().

Please share your experience using stand alone library.

Thanks.

kulaksazov, 2021-09-10:
I had the same problem with my heltec esp 32 LoRa - at any moment it just stopped taking! There is no such thing with a transmitter.
The problem is solved very easily!
In the loop section, or where the LoRa read loop is, after the end of the loop just put: 

**// put the radio into receive mode
        LoRa.receive();**

just to "remind" each time the device that you need to keep taking!
**It hasn't stopped yet! At least for me the problem is solved** :)

fjohannes, 2021-09-20:
For me, adding the yield(); after the line with Lora.parsePacket(); did the trick!
It is working flawlessly and reliable for hours since then.
As my receiver also sends a confirmation message back, setting it to LoRa.receive(); mode completely would not be practical.
btw: I did not have to add the yield() to the sender, receiving the (relatively short) confirmation message [horizontal checksum].

theowlpo, 2022-04-04:
Hi,
I had the same problem as described above and solved the problem as @fjohannes did, being adding yeld() after the line Lora.parsePacket();

livioenrico, 2024-07-20:
I found a solution !!!

Many people have experienced this but until now no one had found a solution:
https://arduino.stackexchange.com/questions/94389/lora-sx1278-stops-receiving-after-a-short-time
https://github.com/sandeepmistry/arduino-LoRa/issues/447
https://github.com/HelTecAutomation/Heltec_ESP32/issues/61
https://stackoverflow.com/questions/63861160/arduino-nano-with-ra-02sx1278-freezes-after-receiving

This is a good workaround, but it wouldn't hurt for SEMTECH to understand why it happens
and also that he would give us the official solution, or fix the defect in his chips.

// ============================================================
//  Each time the OnReceive callback is called there is a small error chance
//  this could be approximately every 1000, 3000 or sometimes 10000 calls. 
// ----------------------------------------------------------------------------
//  When this happens the DIO0 interrupt pin remains stuck high
//  with a voltage fixed to 3.3V and interrupt receiving no longer works.
// ----------------------------------------------------------------------------
//  To correct this problem we test in the loop() if the DIO0_PIN is HIGH
//  and in this case we redo all the initializations.
// ============================================================

// ============================================================
//  RESTART RECEIVER INTERRUPT IF LOCKED
//  This function must be called periodically from the loop()
// ============================================================
void RestartReceiverInterruptIfLocked()
{
  if (digitalRead(DIO0_PIN) == HIGH)
  {
    RTX_Init();
  }
}

// ============================================================
//  RTX INIT
// ============================================================
void RTX_Init()
{
  // ---------------------------------------------- BEGIN
  while (!LoRa.begin(RTX_FREQUENCY))
  { 
    Serial.println("Error initializing LoRa module");
    delay(100);
  }
  // ---------------------------------------------- SETTINGS
  LoRa.setTxPower(17); // do here your additional settings           
  // ---------------------------------------------- REGISTER CALLBACK
  LoRa.onReceive(RTX_OnReceive); 
  // ---------------------------------------------- START RECEIVE 
  LoRa.receive();
}

livioenrico, 2024-07-22:
The yield() solution can be applied only if the code works in polling mode:
```
void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  yield();
   . . .
   . . .  and so on. . . 
```
I do not know if this solution has some effect because working in polling mode my test are always passed without problems. And my test are really long, more than 800 000 TX and RX (ten transmissions per second for 24 hours) without any problem. While doing similar tests please work in isolated room and without antennas.

But the locking problem happens when the code is in interrupt mode (with the DIO0 pin connected) and you register the onReceve() callback in the setup():
```
 // ---------------------------------------------- REGISTER CALLBACK
  LoRa.onReceive(RTX_OnReceive); 
  // ---------------------------------------------- START RECEIVE 
  LoRa.receive();
   . . .
```

gh-user-source, 2024-07-22:
This software blows fat cocks, let it die.

livioenrico, 2024-07-22:
What software are you talking about?
If you're talking about Radiolib.h, I also tried with Sandeep Minstry's LoRa.h and the same error happens. The error is not in the Arduino libraries but in the SX1278 chip. Or maybe there is something we should do that Semtech would do better to explain more clearly in the datasheets.

issacarbrun, 2025-11-22:
I am passing the same issue with lorawan, using the example LoRaWANOled, changing the time to sending in 5 minutes only in ABP mode. 
Maybe this solution works as well?