HelTecAutomation/Heltec_ESP32 #83: Is the demo source code available somewhere?

levycarneiro, 2022-01-20:
I have the Heltec ESP32 LoRa v2 board and it comes preloaded with a demo that shows the available Wifi networks and then transitions to a LoRa sender function.

Is this source code available somewhere? I want to read those packets from another (equal) board, but this one with my own "lora receiver" code.

Thanks.

levycarneiro, 2022-01-23:
After some tests, found that the demo app uses the following settings:

Spreading factor = 11
Sync word = 0x34

closed 2022-01-23 by levycarneiro

IsaacAmante, 2023-06-05:
Where did you find the code? Is it working on arduino or PlatformIO? I'm trying to migrate to PlatformIO but having lots of problems. Um olá do Brasil pra você!

levycarneiro, 2023-06-05:
Hello @IsaacAmante ,

this is where I found the demo and the default values:

https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/src/lora/LoRa.cpp

I never used PlatformIO so can't comment on that.

Abraço!

IsaacAmante, 2023-06-05:
I'm having problems using that library on PlatformIO. On arduino I'm using this example: 
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/esp32/libraries/Heltec-Example/examples/Factory_Test/WiFi_LoRa_32_V2_FactoryTest/WiFi_LoRa_32_V2_FactoryTest.ino
It's pretty much equal to the preinstaled factory test. Thank's!