HelTecAutomation/Heltec_ESP32 #85: error: call of overloaded Wire.begin(int&, int, int)' is ambiguous

platypii, 2022-05-13:
The code in BMP180.cpp does not compile after upstream arduino-esp32 library was updated.

```
.pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/BMP180.cpp:13:29: error: call of overloaded 'begin(int&, int, int)' is ambiguous
   Wire.begin(sda, 12, 100000);
                             ^
In file included from .pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/BMP180.h:5,
                 from .pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/BMP180.cpp:1:
.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:79:10: note: candidate: 'bool TwoWire::begin(int, int, uint32_t)'
     bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
          ^~~~~
.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:80:10: note: candidate: 'bool TwoWire::begin(uint8_t, int, int, uint32_t)'
     bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);
          ^~~~~
```

I'm trying to use this library via platformio. My code worked, and was unchanged for more than 1 year. I just went back to the project, updated library versions, and now I'm getting a compile error from this library.

Specifically, in [BMP180.cpp](https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/src/BMP180.cpp#L12) there is a call to `Wire.begin`.

This used to be fine. But then the underlying `framework-arduinoespressif32` library was [updated](https://github.com/espressif/arduino-esp32/pull/5746) to add an overloaded `Wire.begin` function, which makes the call here ambiguous.

```diff
     bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
+    bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);
```

I tried some changes to the BMP180.cpp in this repo to see if I could fix it, but I couldn't figure out how to make the compiler know which `Wire.begin` function to use. I also left a [comment](https://github.com/espressif/arduino-esp32/pull/5746#discussion_r872019028) on the arduino-esp32 repo to see if they have any suggestions.

platypii, 2022-05-13:
Sorry, just saw this was fixed in commit [494d6b9c](https://github.com/platypii/Heltec_ESP32/commit/494d6b9c9e206ae0515ad833bafe2b9b7dc99c40). Closing.

platypii, 2022-05-13:
Reopening: seems like there hasn't been a release since June 2020, and so I can't use the fix that is already merged.

Any chance of publishing a new version 1.1.1? :pray:

giukio, 2022-05-17:
I agree this fix is necessary.

Compiling .pio\build\heltec_wireless_stick_lite\libcc3\Heltec ESP32 Dev-Boards\heltec.cpp.o
.pio/libdeps/heltec_wireless_stick_lite/Heltec ESP32 Dev-Boards/src/BMP180.cpp: In member function 'boolean BMP085::begin(uint8_t)':
.pio/libdeps/heltec_wireless_stick_lite/Heltec ESP32 Dev-Boards/src/BMP180.cpp:12:28: error: call of overloaded 'begin(int, int, int)' is ambiguous
   Wire.begin(13, 12, 100000);

platypii, 2022-05-17:
Version 1.1.1 was published last night to the Arduino Library by @Heltec-Aaron-Lee. Thank you Aaron!!
https://www.arduino.cc/reference/en/libraries/heltec-esp32-dev-boards/

Still waiting for it to be published to the PlatformIO Registry.
https://registry.platformio.org/libraries/heltecautomation/Heltec%20ESP32%20Dev-Boards

Aaron said: "platformio is maintained by their official team for us, and I have notified them to upgrade accordingly". So I hope it will update soon. :crossed_fingers:

platypii, 2022-05-18:
Package 1.1.1 is [published](https://registry.platformio.org/libraries/heltecautomation/Heltec%20ESP32%20Dev-Boards) to PlatformIO registry! And my project now works with the latest `espressif32` and `Heltec ESP32 Dev-Boards` packages!

THANK YOU HELTEC!! :purple_heart: Awesome customer service

closed 2022-05-18 by platypii

johannes-777, 2022-05-31:
You mean you can compile with the PlatformIO Espressif 32 V4.3.0?
How is this possible?  This issue is still open:
http://community.heltec.cn/t/error-compiling-example-esp32-v2/10327/3
https://github.com/HelTecAutomation/ESP32_LoRaWAN/issues/79

platypii, 2022-05-31:
Yes I am still able to compile with espressif32 4.3.0. Maybe because I'm not actually using LoRaWAN in my project?

johannes-777, 2022-05-31:
You are right.  I just compiled a minimal project with espressrif V4.3 as well.
My impression is, that heltec has pretty much deserted its roots, i.e. the boards that support LoRaWAN.  I am trying to get LMIC running directly (https://github.com/mcci-catena/arduino-lmic).
To my surprise, there is already a config for Heltec boards:
https://github.com/mcci-catena/arduino-lmic/tree/master/src/hal