HelTecAutomation/Heltec_ESP32 #95: A fatal error occurred: Invalid head of packet (0x47).  Wireless stick lite (V3)

ccintront, 2022-11-05:
Hello, I just received Heltec wireless stick lite (v3) [https://heltec.org/project/wireless-stick-lite-v2/] module. The code is not getting uploaded on the board. I have tried different sample codes. 

Wireless stick lite V3 board
[image]

**Whenever I try to upload the code, I get this error**

[image]

I have worked with the previous version of Heltec wireless stick lite, and it worked completely fine and uploading does not cause any issue, but V3 is having some problems in uploading code.

I bought 5 of these boards, and it is not working with any of them.

Any suggestions?

docbrownd, 2022-11-27:
> Hello, I just received Heltec wireless stick lite (v3) [https://heltec.org/project/wireless-stick-lite-v2/] module. The code is not getting uploaded on the board. I have tried different sample [...]

same here, did you find a solution ?

ccintront, 2022-11-27:
Not yet. Still the same.

docbrownd, 2022-11-27:
> Not yet. Still the same.

for me it's because we use wrong board version. 

Did you try with the new heltec version (0.0.7) https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.7/package_heltec_esp32_index.json ? 

with it, i think programmation will work, but in my case, compilation fails with V2 code (LoRa var not declared ..)

GingerJohnny, 2023-01-22:
The new V3 board chip i different. I am experiencing a problem uploading via VS Code whiere-in I get the fatal upload error that the chip is an ESP32-S3 not and ESP32. I have submitted an issue here as well. We'll see wht hsppens.

GingerJohnny, 2023-01-22:
Here is the V3 chip diagram from Heltec http://community.heltec.cn/t/heltec-esp-32-v3-adruino-sketch/12207

sensei-hacker, 2023-02-11:
For the V3 we need to new versions of the board and library.
Close the Arduino IDE.  Delete the old version, which will be in a directory such as ~/.arduino15/packages/Heltec-esp32/hardware/esp32/0.0.5

Open the Arudino IDE, then select File > Preferences and set this URL to get version 0.0.7 of the library:
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.7/package_heltec_esp32_index.json

Note that version 0.0.5 won't work. You need 0.0.7 or higher.

Go to boards manager. Update Heltec ESP32 to 0.0.7 or higher.

Click Tools > Board and select the V3 version of your board.

Now ESP32 examples should work.

If you're doing LORA, you may need to download [the HelTec ESP32_LoRaWAN library](https://github.com/HelTecAutomation/ESP32_LoRaWAN/archive/master.zip) find into the directory below and unzip.

linux: /home/{user}/Arduino/libraries
windows: Documents\Arduino\libraries
mac os: Documents/Arduino/libraries

bigFin, 2023-02-28:
@sensei-hacker incorrect about LoRa. The library is incompatible with SX1262 ([source](https://github.com/sandeepmistry/arduino-LoRa/blob/master/README.md#compatible-hardware) )

#94 #100 #102

bigFin, 2023-02-28:
I am able to use the SX1262 with radiolib similar to this example [https://github.com/JJJS777/Hello_World_RadioLib_Heltec-V3_SX1262](https://github.com/HelTecAutomation/Heltec_ESP32/issues/Hello_World_RadioLib_Heltec-V3_SX1262)

garykuipers, 2023-06-19:
> For the V3 we need to new versions of the board and library. Close the Arduino IDE. Delete the old version, which will be in a directory such as [...]

I am compliant with the above, neverhteless:

Archiving built core (caching) in: /tmp/arduino_cache_663687/core/core_esp32_esp32_heltec_wireless_stick_PSRAM_disabled,CPUFreq_240,UploadSpeed_921600,DebugLevel_none,LORAWAN_REGION_3,LoRaWanDebugLevel_0,EraseFlash_none_620a1fac7f13b6493682adf515800eea.a
Sketch uses 744033 bytes (22%) of program storage space. Maximum is 3342336 bytes.
Global variables use 43796 bytes (13%) of dynamic memory, leaving 283884 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting...

A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?
A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?

garykuipers, 2023-06-19:
I have a Heltec Wirelss Stick:
The options are Heltec Wireless Stick: When chosen it complains it is the wrong board and suggests (V3)

When the board selected is the V3 version it does not have a Wireless Stick, only a Wireless Stick Lite (supposedly obsolete)

When trying to compile the factory test example:

heltec_wireless_stick:59:10: error: 'class Heltec_ESP32' has no member named 'display'
   Heltec.display->clear();

It would be REALLY REALLY NICE, if some day, I could buy a new Heltec product and NOT SPEND DAYS TRYING TO GET IT TO WORK! Trust me, I would buy more if they worked out of the box rather than casuing days of pain.

bigFin, 2023-06-19:
@garykuipers Oh the joy of using the latest hardware... You will be able to compile using the following:

```
platform = espressif32
board = heltec_wifi_lora_32_V3
```
Just note that you will need to implement LoRa libraries etc yourself (see my above post about RadioLib).