HelTecAutomation/Heltec_ESP32 #96: Software Support for the new Wifi Kit 32 (V3) is horrible

wiegleyj, 2022-11-13:
Factory delivered program is non-existent. Examples for OLED drivers reference SDA_OLED which is not defined anywhere in any files in the repository. esp-idf support is non-existent. The software support for these devices is incredibly confusing and disorganized.

panamantis, 2022-12-06:
Wow.  It sounds like having a wiki to organize this incoming issues may make sense?  I saw the aliexpress pages warn that it's only for the weary at the moment.  And, that it only supports esp-idf for now?  I've got a bunch coming via fedex and plan to plow through issues.

spleenware, 2022-12-08:
Wow, have to agree about the lack of documentation. It's woeful.

Baptou88, 2022-12-08:
indeed, there is a complete lack of information, which can make development extremely difficult and slow for beginners (I am one of those people).  I recently bought the new V3 boards but I noticed that these new boards are not in platformIo nor in esp-idf, so I tried to fork platformIo and esp-idf which I  used to develop

nossonCotlar, 2023-01-12:
I am disappointed with this board too - bought 4 of them without knowing what I was getting into. Any material I could find on it was posted within the last few days. I am a beginner to this stuff too. 
To even get platformio to build and upload was a goose chase, needing to point the platform to a farked version of arduinoespressif32 guided by the conversation I found here https://community.platformio.org/t/heltec-board-setup-libraries-esp32-s3-wifi-kit-32-v3-w-oled-ssd1306/31316
Anyway, I finally found the right pins by tracking them down thru the installed library from the fork
https://github.com/espressif/arduino-esp32/blob/d342739308436e13e56a5c4d3fe597076a1d36b3/variants/heltec_wifi_kit_32_v3/pins_arduino.h#L71-L73
```
RST_OLED = 21
SCL_OLED = 18
SDA_OLED = 17
```
Hope this is useful to others, because I can't find this anywhere else on the internet at the moment. The datasheet for wifi_kit_32_v3 is also complete canine-doodoo and only mentions the reset pin. Thank God these pins finally work for my v3.

tags so others can google #heltec #wifi_kit_32_v3 #esp32-s3

sensei-hacker, 2023-02-11:
For the V3 we need to new versions of the board and library.
Close the Arduino IDE.  Delete the old version, which will be in a directory such as ~/.arduino15/packages/Heltec-esp32/hardware/esp32/0.0.5

Open the Arudino IDE, then select File > Preferences and set this URL to get version 0.0.7 of the library:
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/download/0.0.7/package_heltec_esp32_index.json

Note that version 0.0.5 won't work. You need 0.0.7 or higher.

Go to boards manager. Update Heltec ESP32 to 0.0.7 or higher.

Click Tools > Board and select the V3 version of your board.

Now ESP32 examples should work.

spadab, 2023-02-13:
> RST_OLED = 21
> SCL_OLED = 18
> SDA_OLED = 17

gasagna, 2023-02-18:
Worst of all is the complete lack of interaction with Heltec. Broken links, incomplete or wrong documentation, etc. This is not good customer support Heltec!

PR #100 (2023-04-26): Support heltec V3 board capabilities
This PR adds support for the latest V3 boards: `WIFI_Kit_32_V3` and `WIFI_LoRa_32_V3`

Without this, things like the screen, wifi, and lora do not work!
Because `heltec.cpp` checks for specific board macros.

Since Heltec now has a lot of boards in their inventory, I refactored `heltec.h` to map each board to certain capabilities: `Heltec_Wifi`, `Heltec_LoRa`, `Heltec_Screen`, `Heltec_Vext`

Then I changed the code in `heltec.cpp` to check for capabilities, not specific boards.
This will make it much easier to release new boards in the future.

Should fix # [... cut]
closed 2023-04-26 by Heltec-Aaron-Lee