HelTecAutomation/Heltec_ESP32 #98: Heltec Wireless Stick Lite V3 Low Power Example Doesn't compile

vleroylr, 2023-02-13:
Apparently there is no _#define_ for the wireless stick lite V3, thus, in the file _Heltec.cpp_ the LoRa object is not created, resulting in a compile error due to the object not existing. Even after adding the #define before the #include "Heltec....." the code compiles but the **current conpsumption is 19mA, WAAAAY more than the stated 15uA for this product.**

mahesh2000, 2023-02-14:
is there *any* documentation or are there any examples for this particular device? it's different enough from the Wireless Stick Lite V2. the deep sleep doesn't compile and, separately, i can't even see any labeled I2C pins.

JJJS777, 2023-02-14:
@mahesh2000 u can find few things on [Heltec](https://heltec.org/project/wifi-lora-32-v3/) under Datasheet or Docs & Resource. I have found the I2C pin in the [Schematic Diagram](https://resource.heltec.cn/download/WiFi_LoRa32_V3/HTIT-WB32LA(F)_V3_Schematic_Diagram.pdf).
I made a litte [_Hello World_](https://github.com/JJJS777/Hello_World_RadioLib_Heltec-V3_SX1262) example with RadioLib on the V3. There is also an example for the OLED. haven't testet deep sleep though.

mahesh2000, 2023-02-16:
There is an entirely different board definition repo on GitHub which you need to install. There are a few ways to do it, none of them simple. I did my own before i saw the others'. See https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/165.

vleroylr, 2023-02-17:
@mahesh2000 Did you get the stated current consumption on low power mode?

mahesh2000, 2023-02-18:
@vleroylr, i got it down to 13uA.

mahesh2000, 2023-02-18:
Also, just use the predefined SDA and SCL pins. You can just use `Serial.println(SDA)` and `Serial.println(SCL)` to find the values. If I remember correctly, for the WSL V3 they're 17 & 18, ie GPIO_NUM_17 and GPIO_NUM_18.

vleroylr, 2023-02-28:
@mahesh2000 Could you share the code where you got it down to 13uA?, I am really struggling to make it work and every help would be much appreciated :( !

vleroylr, 2023-05-16:
@mahesh2000 , Hey friend, I wonder if you could share your implementation on the  Wireless Stick Lite V3 on low power mode... I still can't get it to 13uA as you did :(

vleroylr, 2023-05-16:
After some tweaking I got it down to 0.65mA, still trying to figure out were are the extra uA comming from. Such a pitty the company won't involve in this issue.