HelTecAutomation/Heltec_ESP32 #105: Heltec ESP Lora32 V3 keeps crashing

plonxyz, 2023-03-23:
Hi 
I tried to use an Lora Example provided by heltech on arduino IDE. 

Somehow my Boards keep crashing with a checksum error:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
pro cpu reset by JTAG
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8

Does anybody know how I can fix this issue?!

Thanks!

bitrot-alpha, 2023-04-03:
see my comment on #106 for how to fix this on the V3 board.
You need to download a new version of `esptool` (program that flashes the ESP32). The one provided by this library is too old and doesn't support the newer ESP32-S3 that comes on the V3 boards.

jonnytest1, 2023-04-24:
maybe also try other examples (for example i tried display api and the first exampel i looked at crashed but the one that interfaced directly with SSD1306Wire worked )

bitrot-alpha, 2023-06-08:
I haven't been able to determine the exact cause of the checksum error, other than that Heltech seems to be using `esptool` somewhat improperly.

For anyone else wondering about this, take a look at Espressif's "Flashing Firmware" page. There's a good note there about how to make the Arduino IDE show the command it's using for `esptool`
https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html

plonxyz, 2023-06-09:
sorry @Shaedo , I never tried it ..