HelTecAutomation/Heltec_ESP32 #106: OLED_LoRa Example sketch compilation error - WiFi LoRa 32 (V3)

TomB4321, 2023-03-28:
I've just received my WiFi LoRa 32 (V3) 415~510 and I am getting errors when I try to compile the example LoRa send/receive examples.

I'm using:
Arduino IDE 2.0.4 on Windows 11
Board library - Heltec ESP32 Series Dev-boards V0.0.7
Library - Heltec ESP32 Dev-Boards v 1.1.1
Example sketches used - OLED_LoRa_Receiver & Sender

This is a new laptop and the only libraries I have installed are for this board..also doesn't work on my old computer same errors.

When I complie the sketch I get the following error -

`C:\Users\helic\AppData\Local\Temp\.arduinoIDE-unsaved2023228-22564-1dwcqvz.bk08\OLED_LoRa_Sender\OLED_LoRa_Sender.ino: In function 'void loop()':
C:\Users\helic\AppData\Local\Temp\.arduinoIDE-unsaved2023228-22564-1dwcqvz.bk08\OLED_LoRa_Sender\OLED_LoRa_Sender.ino:66:3: error: 'LoRa' was not declared in this scope
   LoRa.beginPacket();
   ^~~~
C:\Users\helic\AppData\Local\Temp\.arduinoIDE-unsaved2023228-22564-1dwcqvz.bk08\OLED_LoRa_Sender\OLED_LoRa_Sender.ino:75:22: error: 'RF_PACONFIG_PASELECT_PABOOST' was not declared in this scope
   LoRa.setTxPower(14,RF_PACONFIG_PASELECT_PABOOST);
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "heltec.h"
  Used: F:\Arduino\libraries\Heltec_ESP32_Dev-Boards
  Not used: C:\Users\helic\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\libraries\Heltec-Example
exit status 1

Compilation error: 'LoRa' was not declared in this scope`

This is my first LoRa board and I don't know enough to trouble shoot this myself and was hoping to use the examples as an aid to learn. I added the LoRa.h to the sketch and the sketch compiles, but the sender board doesn't appear to send anything (the counter doesn't increment) and the receiving board doesn't get a signal. I've seen from a couple of YouTube videos other people testing these sketches, but with a V2 board, and I haven't done anything different to them....

Any help on this issue would be greatly appreciated!

bitrot-alpha, 2023-04-03:
Don't use this library (it's outdated or something), I don't know the extent of the issues but it doesn't work out of the box. Just use the Aaron Lee library that you already have installed - https://docs.heltec.org/en/node/esp32/quick_start.html#via-arduino-board-manager

For you @TomB4321 just uninstall the Heltec ESP32 library (not the BOARD library, this library). You can do that by opening the library manager and hitting the uninstall button on the library (https://support.arduino.cc/hc/en-us/articles/360016077340-Uninstall-libraries-from-Arduino-IDE ). The code should compile and run fine.

For people starting from scratch:

Issue - If you're using a WiFi LoRa **V3** board, with the ESP32-S3, the chip constantly resets after programming it.
You'll have to replace the `esptool` program (issue details at https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/159 ) with one >= 4.2 (download at https://github.com/espressif/esptool/releases/latest ).

Step by step:
1. Follow https://docs.heltec.org/en/node/esp32/quick_start.html#via-arduino-board-manager
2. Open the Arduino15 folder - https://support.arduino.cc/hc/en-us/articles/360018448279-Open-the-Arduino15-folder
3. Download latest `esptool` https://github.com/espressif/esptool/releases/latest (look for `win-64` .zip file)
4. Replace `esptool` in C:\Users\YOUR_USER\AppData\Local\Arduino15\packages\Heltec-esp32\tools\esptool_py\3.3.0\ with the one you just downloaded. Make sure to extract the zip file - you need all of the files inside of it.

gchoquet, 2023-04-05:
@bitrot-alpha, thank you. It solved the same problem for me. 

I'll have check to see if it still works for the V2 boards I have. 

It appears to me the problem is in the associated example programs not having the correct header files.

It would be really nice if Hiltec's instructions were correct.

You were a lifesaver, Thank you for the assistance.

deeprotype, 2023-04-28:
@jonnytest1 your are right!

Had the same issue - Probleme has been solved by adding #define WIFI_LoRa_32 before adding heltec.h.

fulviomanente, 2023-04-28:
I went the other way around guys. As they had already made all the setup to control de Oled inside the new LoraWan_APP library, instead of fighting it I removed all references to halted.h and implemented these 2 new methods on the lib so I can write to the Oled like we used to do on version 2:

on LoraWan_APP.cpp

void LoRaWanClass::displayTxt(uint8_t line, char *message)
{
        isDispayOn = 1;
        digitalWrite(Vext,LOW);
        display.drawString(0, line, message);     
        display.display();
        delay(500);
}
void LoRaWanClass::displayClean()
{
        display.clear();
}
void LoRaWanClass::displayInit()
{
        display.init();
	display.setFont(ArialMT_Plain_10);
        display.setTextAlignment(TEXT_ALIGN_LEFT); 
} 
#endif

on LoraWanAPP.h

#if defined(WIFI_LoRa_32_V3)||defined(WIFI_LoRa_32_V2)
  void displayJoining();
  void displayJoined();
  void displaySending();
  void displayAck();
  void displayMcuInit();
  void displayTxt(uint8_t line, char *message);
  void displayClean();
  void displayInit();
#endif
};

(BOTTOM 3 ARE THE NEW METHODS)

I cannot upload it here but I can share with you the full files if it helps. I didn't get to the point to have a generic code for V2 and V3 but it's all compatible with V3 thinking about the future

EdAChu, 2024-07-06:
Does anyone know where to find the WiFi_LoRa_32_V3_FactoryTest ino file?

jonnytest1, 2024-07-07:
> Does anyone know where to find the WiFi_LoRa_32_V3_FactoryTest ino file?

https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/examples/Factory_Test/WiFi_LoRa_32_V3/WiFi_LoRa_32_V3_FactoryTest_V1/WiFi_LoRa_32_V3_FactoryTest_V1.ino u cna just search for files in the repo (thers also a V2)