HelTecAutomation/Heltec_ESP32 #124: Most examples fail to compile

autotel, 2023-09-12:
Most examples use `#include` "LoRaWan_APP.h" however, this library does not come with that file. I have not been able to get any example to compile, except for the "factory test"

example: https://github.com/HelTecAutomation/Heltec_ESP32/blob/9b066f5166ed15abe612c81343bd624b12d55906/examples/LoRa/LoRaReceiver/LoRaReceiver.ino#L1C22-L1C22

MuhammedZek, 2023-09-14:
that might help you https://www.aeq-web.com/heltec-lora32-v3-board-arduino-ide-lorawan-setup/
you need to install the right boad. if you use ESP32 dev bord it wont work

autotel, 2023-09-14:
thank you very much, @MuhammedZek 
I was in a rush, so instead I imagined that this was a custom board, and thus I sought and installed separate compatible libraries for the screen, and the lora wan. That approach worked well :)

Mr-HaleYa, 2024-01-10:
@autotel 
If your issue has been solved please close