HelTecAutomation/Heltec_ESP32 #132: "HELTEC_BOARD was not declared in this scope" - LoRaSender.ino

Frosty-STI, 2024-03-26:
I receive these errors upon compiling LoRaSender.ino: 

C:\Users\brend\OneDrive\Documents\Arduino\LoRaReceiver\LoRaReceiver.ino: In function 'void setup()':
C:\Users\brend\OneDrive\Documents\Arduino\LoRaReceiver\LoRaReceiver.ino:57:15: error: 'HELTEC_BOARD' was not declared in this scope
     Mcu.begin(HELTEC_BOARD,SLOW_CLK_TPYE);
               ^~~~~~~~~~~~
C:\Users\brend\OneDrive\Documents\Arduino\LoRaReceiver\LoRaReceiver.ino:57:28: error: 'SLOW_CLK_TPYE' was not declared in this scope
     Mcu.begin(HELTEC_BOARD,SLOW_CLK_TPYE);
                            ^~~~~~~~~~~~~

exit status 1

Compilation error: 'HELTEC_BOARD' was not declared in this scope

Board:
Heltec HiLetgo ESP32 Wireless Stick (SX1276)

I have installed the Heltec_ESP32 library, added the board URL to the board manager ( https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series.git ), and selected 'Heltec Wireless Stick' as my target device.

GM-BLUE13, 2024-03-27:
I have the same issue. Did you find something?

Mr-HaleYa, 2024-03-27:
You added `https://resource.heltec.cn/download/package_heltec_esp32_index.json` to your board manager, restarted Arduino, and installed the board and it looks like this?
[image]

I had that error when I had the old board framework still installed

Frosty-STI, 2024-03-30:
Thank you @Mr-HaleYa !!!!!!!!!!! Works perfect now. I wasn't sure if I had the right board manager URL but the one you provided definitely is, thanks again :)

closed 2024-03-30 by Frosty-STI