HelTecAutomation/Heltec_ESP32 #134: 2.0.1 Breaks Oled

valdt, 2024-03-28:
The lora exemples works but the oledexemples dose not work. And vice versa if you down grade to 1.1.5

c:/users/valdt/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\valdt\AppData\Local\Temp\arduino\sketches\31CA39F4CB3EC676FA1798AC4D335946\libraries\Heltec_ESP32_Dev-Boards\LoRaWan_APP.cpp.o:c:\Users\valdt\OneDrive\Dokument\Arduino\libraries\Heltec_ESP32_Dev-Boards\src/LoRaWan_APP.cpp:28: multiple definition of `display'; C:\Users\valdt\AppData\Local\Temp\arduino\sketches\31CA39F4CB3EC676FA1798AC4D335946\sketch\OledTest.ino.cpp.o:C:\Users\valdt\OneDrive\Dokument\GitHub\Heltec-Lora-Test\OledTest/OledTest.ino:6: first defined here
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

valdt, 2024-03-29:
#include "heltec.h"
#include "Arduino.h"

//rotate only for GEOMETRY_128_64
 SSD1306Wire display(0x3c, SDA_OLED, SCL_OLED, RST_OLED);

void VextON(void)
{
  pinMode(Vext,OUTPUT);
  digitalWrite(Vext, LOW);
}

void VextOFF(void) //Vext default OFF
{
  pinMode(Vext,OUTPUT);
  digitalWrite(Vext, HIGH);
}

void setup() {

  VextON();
  delay(100);

  display.init();
  display.clear();
  display.display();
  
  display.setContrast(255);
  
  display.setTextAlignment(TEXT_ALIGN_CENTER);
  display.clear();
  display.display();
  display.screenRotate(ANGLE_0_DEGREE);
  display.setFont(ArialMT_Plain_16);
  display.drawString(64, 32-16/2, "ROTATE_0");
  display.display();
  delay(2000);

  display.clear();
  display.display();
  display.screenRotate(ANGLE_90_DEGREE);
  display.setFont(ArialMT_Plain_10);
  display.drawString(32, 64-10/2, "ROTATE_90");
  display.display();
  delay(2000);

  display.clear();
  display.display();
  display.screenRotate(ANGLE_180_DEGREE);
  display.setFont(ArialMT_Plain_16);
  display.drawString(64, 32-16/2, "ROTATE_180");
  display.display();
  delay(2000);

  display.clear();
  display.display();
  display.screenRotate(ANGLE_270_DEGREE);
  display.setFont(ArialMT_Plain_10);
  display.drawString(32, 64-10/2, "ROTATE_270");
  display.display();
  delay(2000);
}

void loop() { }

joelkoz, 2024-04-15:
Is this related to #137 ?

Jah-On, 2024-04-19:
Can confirm.

bassklampfe, 2025-03-04:
I found a reason:
In heltec.h the is a block
```c++
#if defined( WIFI_LORA_32 ) || defined( WIFI_LORA_32_V2 ) || defined( WIFI_LORA_32_V3 )
#define Class_WIFI_LORA
#endif
```
In heltec.cpp there is a block
```c++
#if defined( Class_Wifi_Kit ) || defined( Class_Wifi_LoRa )
	display = new SSD1306Wire(0x3c, SDA_OLED, SCL_OLED, RST_OLED, GEOMETRY_128_64);
#elif defined( WIRELESS_STICK )
	display = new SSD1306Wire(0x3c, SDA_OLED, SCL_OLED, RST_OLED, GEOMETRY_64_32);
#endif
```
The definition and the usage simply does not match in case. This `display` is never initialized and first access results in a nullptr

Jah-On, 2025-03-06:
I use the [unofficial Heltec library](https://github.com/ropg/heltec_esp32_lora_v3) the this official one is a joke.