HelTecAutomation/Heltec_ESP32 #137: OLED display not initialized correctly for wifi_kit or Wifi_LoRa

joelkoz, 2024-04-15:
I am trying to compile using the latest source for a wifi_kit_32_v3 board and the OLED does not work. I noticed that "heltec.cpp" has the following line 11:

`
#if defined( Class_Wifi_Kit ) || defined( Class_Wifi_LoRa )
display = new SSD1306Wire(0x3c, SDA_OLED, SCL_OLED, RST_OLED, GEOMETRY_128_64);
`

yet the new SSD1306Wire class has a constructor with signature:

`
SSD1306Wire(uint8_t _address, uint32_t _freq, int sda,int scl, DISPLAY_GEOMETRY g = GEOMETRY_128_64,int8_t _rst=-1)
`

I believe the line in heltec.cpp needs to be:

`
display = new SSD1306Wire(0x3c, 700000U, SDA_OLED, SCL_OLED, GEOMETRY_128_64, RST_OLED);
`

joelkoz, 2024-04-15:
Line 13 of heltec.cpp appears to have the wrong signature also. This issue may appear elsewhere. I do not have a wireless stick to test, so I am not making a PR, but I'm pretty sure those two lines both need to be fixed for the OLED displays to work.

bassklampfe, 2025-03-04:
There is an other issue

In heltec.h the is a block
```c++
#if defined( WIFI_LORA_32 ) || defined( WIFI_LORA_32_V2 ) || defined( WIFI_LORA_32_V3 )
#define Class_WIFI_LORA
#endif
```
In heltec.cpp there is a block
```c++
#if defined( Class_Wifi_Kit ) || defined( Class_Wifi_LoRa )
	display = new SSD1306Wire(0x3c, SDA_OLED, SCL_OLED, RST_OLED, GEOMETRY_128_64);
#elif defined( WIRELESS_STICK )
	display = new SSD1306Wire(0x3c, SDA_OLED, SCL_OLED, RST_OLED, GEOMETRY_64_32);
#endif
```
The definition `Class_WIFI_LORA` and the usage `Class_Wifi_LoRa` simply does not match in case. This display is never initialized and first access results in a nullptr