HelTecAutomation/Heltec_ESP32 #139: Numerous compiler errors since 2.0.1 update

G6EJD, 2024-04-18:
Library Heltec ESP32 Dev-Boards has been declared precompiled:
Using precompiled library in d:\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\esp32s3
d:\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\LoRaWan_APP.cpp: In member function 'void LoRaWanClass::sleep(DeviceClass_t)':
d:\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\LoRaWan_APP.cpp:666:33: error: 'HELTEC_BOARD' was not declared in this scope
  Mcu.sleep(classMode,debugLevel,HELTEC_BOARD,SLOW_CLK_TPYE);
                                 ^~~~~~~~~~~~
d:\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\LoRaWan_APP.cpp:666:46: error: 'SLOW_CLK_TPYE' was not declared in this scope
  Mcu.sleep(classMode,debugLevel,HELTEC_BOARD,SLOW_CLK_TPYE);
                                              ^~~~~~~~~~~~~
Multiple libraries were found for "LoRaWan_APP.h"
  Used: D:\Documents\Arduino\libraries\Heltec_ESP32_Dev-Boards
  Not used: C:\Users\David\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\1.0.0\libraries\LoraWan102
exit status 1

Compilation error: exit status 1

lapinos03, 2024-04-24:
I had same issue when trying to compile Files -> Examples -> Heltec ESP32 Dev-boards -> Factory_Test -> WiFi_LoRa_32_V3 -> WiFi_LoRa_32_V3_FactoryTest_V1.

I had chosen `Heltec WiFi LoRa 32(V3)` from the Board Manager, which is wrong (this board is defined in Espressif's package, not Heltec's one). One must better choose from the menu **Tools -> Board: XXX -> Heltec ESP32 Series Dev-boards -> \<the board you need\>**. Then, macros `HELTEC_BOARD` and `SLOW_CLK_TPYE` will be defined (it is actually done in file [...]/Arduino15/packages/Heltec-esp32/hardware/esp32/3.0.0/boards.txt, under entry `<your_board>.build.defines`).

G6EJD, 2024-04-24:
I’ll try that, thanks for the help.

G6EJD, 2024-04-25:
The recent changes to the IDE have now resolved the issue.

closed 2024-04-25 by G6EJD