HelTecAutomation/Heltec_ESP32 #145: Cannot compile WiFi_LoRa_32_V3_FactoryTest_V2

romoloman, 2024-05-07:
During linking the linker raise the following error:

 undefined reference to `ex_32k_start_cnt'

I have installed the latest HELTEC ESP32 development framework following the instructions in https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/InstallGuide/windows.md and the latest 2.0.1 library using Arduino 2.3.2 library manager.

where is the ex_32k_start_cnt variable defined ?

If I remove the following part in setup() the example links correctly:

 if(ex_32k_start_cnt>5)
  {
    VextON();
    delay(100);
    factory_display.init();
    factory_display.setFont(ArialMT_Plain_16);
    factory_display.setTextAlignment(TEXT_ALIGN_CENTER);
    packet ="EX 32K ERROR";
    factory_display.clear();
    factory_display.drawString(64, 24, packet);
    factory_display.display();
    while(1);
  }

what is ex_32k_start_cnt supposed to count or do ?

Best regards

rodrigobutta, 2024-05-07:
After somme hours of digging, I´ve found this Issue that explains that, it appears that indeed was an error in the Heltec ESP32 that they fixed on Mar 27 (while the last release is from MMar 22), but Heltec has still not published the release. For others, downgrading worked, not for me, at least not with the `LoRaWan_APP` library. So I hope they publish this soon (more than a month already, sad).

https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/248

MiningPickaxe, 2024-07-02:
The ex_32k_start_cnt seems to be linked to an external 32K RTC.
To me it looks like the code is trying to check if the RTC is working properly. The variable might count something like start attempts or similar. If the RTC didn't initialize/start properly, it shows text on the screen "EX 32K ERROR".

I don't see a RTC chip in the schematic published by Heltec for the v3.1, so I assume this code has been added for some internal test unit or maybe a future revision which includes a dedicated rtc. 

It should be safe to remove the 
```  if(ex_32k_start_cnt>5)
  {
    VextON();
    delay(100);
    factory_display.init();
    factory_display.setFont(ArialMT_Plain_16);
    factory_display.setTextAlignment(TEXT_ALIGN_CENTER);
    packet ="EX 32K ERROR";
    factory_display.clear();
    factory_display.drawString(64, 24, packet);
    factory_display.display();
    while(1);
  }
  ```
  to get it to compile. All it does is output an error in some edge case during startup.

roberthadow, 2024-07-09:
My problem is simpler: 'class WiFiClass' has no member named 'setAutoConnect'

```
Multiple libraries were found for "WiFi.h"
 Used: /home/rhadow/snap/arduino/85/.arduino15/packages/esp32/hardware/esp32/3.0.2/libraries/WiFi
 Not used: /snap/arduino/85/libraries/WiFi
Using library WiFi at version 2.0.0 in folder: /home/rhadow/snap/arduino/85/.arduino15/packages/esp32/hardware/esp32/3.0.2/libraries/WiFi 
Using library Network at version 1.0.0 in folder: /home/rhadow/snap/arduino/85/.arduino15/packages/esp32/hardware/esp32/3.0.2/libraries/Network 
Using library Heltec_ESP32_Dev-Boards at version 2.0.2 in folder: /home/rhadow/snap/arduino/current/Arduino/libraries/Heltec_ESP32_Dev-Boards 
Using library SPI at version 2.0.0 in folder: /home/rhadow/snap/arduino/85/.arduino15/packages/esp32/hardware/esp32/3.0.2/libraries/SPI 
Using library Wire at version 2.0.0 in folder: /home/rhadow/snap/arduino/85/.arduino15/packages/esp32/hardware/esp32/3.0.2/libraries/Wire 

```
Question (1): How do I point the compiler to the correct WiFi.h so this works?
Question (2): How do I stop the compiler from looking in all the wrong places?

matthewbullweb, 2024-08-27:
Hi Had the same issues tonight. V1 works fine but V2 won't. I dumped the original fw before so I can put back. Doesn't help complie though.

VeniceInventors, 2024-09-20:
> Hi Had the same issues tonight. V1 works fine but V2 won't. I dumped the original fw before so I can put back. Doesn't help complie though.

How do you dump the firmware?

I have a slightly different issue, trying to compile the factory test for the Lora32_V3 before replacing it with my own code, and getting errors where a bunch of constants are undefined:
```
  pinMode(RADIO_DIO_1,ANALOG);
  pinMode(RADIO_NSS,ANALOG);
  pinMode(RADIO_RESET,ANALOG);
  pinMode(RADIO_BUSY,ANALOG);
  pinMode(LORA_CLK,ANALOG);
  pinMode(LORA_MISO,ANALOG);
  pinMode(LORA_MOSI,ANALOG);
```
I can probably fix the errors by defining the constants according to the pins_arduino.h file but I'd still like to have a copy of the current firmware until I figure out where the library is broken.

eabase, 2024-11-25:
BUMP! 
I get:
`\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\HT_st7789spi.h:30:10: fatal error: Adafruit_GFX.h: No such file or directory`

matthewbullweb, 2024-11-25:
@eabase You probably haven't got all the framework files installed. Heltec have their own hardware framework for use with their library. There setup guide is here. https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/InstallGuide/windows.md

eabase, 2024-11-25:
@matthewbullweb 
Hi Matthew,
That is the problem. 

1. The link of that page is dead:  
   [How to install Git and Arduino IDE](https://heltec-automation-docs.readthedocs.io/en/latest/general/how_to_install_git_and_arduino.html)
   
2. I don't have any directory called:  
   `[YOUR_USER_NAME]/Documents/Arduino/hardware/heltec`

eabase, 2024-11-25:
```bash
Compilation error: no matching function for call to 'SSD1306Wire::SSD1306Wire(int, int, gpio_num_t, gpio_num_t, OLEDDISPLAY_GEOMETRY, gpio_num_t)'
```

matthewbullweb, 2024-11-26:
@eabase Ignore the dead link and follow rest of instructions on page. E.g. git clone commands, and get.exe in hardware folder (create that is it doesn't exist). You should then have another set of esp32 helteh boards to choose from.

eabase, 2024-11-26:
@matthewbullweb 
Thank you Matthew! 

I had to read the instructions from there and [here](https://docs.heltec.org/en/node/esp32/esp32_general_docs/quick_start.html), probably a good 6 times, backwards and forwards. I couldn't get it to work from within the Arduino-IDE (!), even after several restarts and wiping firewall settings. 

At the end, what got me in port was using Cygwin and Powershell, and ultimately running the python version of `get` from that directory. (Also, it's not funny to see that almost every Windows executable (`*.exe`) coming from Heltec or Arduino are malware flagged with several trojans in _Virus Total_. 👎 

After a lot of wasted time, I really wish this "out-of-box" experience would have been more clearly documented, or at least updated.

**The Compilation**

They really have to update the example code! 

This is quite disturbing as it is very fragile, since there are:
- at least 4 redefinition's of the constants: `LORA_MAC_PUBLIC_SYNCWORD` and `LORA_MAC_PRIVATE_SYNCWORD`
- Dozens of `deprecation` warnings.
- 'unsigned int' warning.

Finally, the compilation is painfully slow! 💤 
Probably because it *seem* to be compiling all libraries also for things not used, like for all world *regions* (~9). For example compiling for China: `..\loramac\region\RegionCN470.c`. 

Any idea how to optimize or speed it up?

```makefile
# From: ..\Arduino15\packages\Heltec-esp32\hardware\esp32\3.0.2\boards.txt

heltec_wifi_lora_32.build.defines=-D{build.band} -DMCU_ESP32_D0 -DWIFI_LORA_32 -DHELTEC_BOARD=1 -DRADIO_CHIP_SX127X -DSLOW_CLK_TPYE=0 -DLoRaWAN_DEBUG_LEVEL={build.LoRaWanDebugLevel} -DACTIVE_REGION=LORAMAC_{build.band} -DLORAWAN_PREAMBLE_LENGTH={build.LORAWAN_PREAMBLE_LENGTH}  -DLORAWAN_DEVEUI_AUTO={build.LORAWAN_DEVEUI_AUTO}  {build.psram}
```

jamesgbahr, 2024-12-04:
> The working method:
> 
> ```shell
> #cd [YOUR_USER_NAME]/Documents/Arduino/hardware/heltec
> cd ~/Documents/Arduino/
> 
> cd C:\Users\$env:USERNAME\Documents\Arduino
> mkdir hardware/heltec
> cd hardware/heltec
> 
> #git clone https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series.git esp32
> git clone https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series.git esp32 --depth 1
> cd esp32
> #? git pull
> git submodule update --init --recursive
> 
> 
> # After this you will find the following files and directories:
> # ll
> 
> Mode                 LastWriteTime         Length Name
> ----                 -------------         ------ ----
> d----          2024-11-25    23:46                pre-commit
> d----          2024-11-25    23:46                partitions
> d----          2024-11-25    23:46                ide-debug
> 
> -a---          2024-11-25    23:46        7018496 get.exe
> -a---          2024-11-25    23:46        6006318 gen_insights_package.exe
> -a---          2024-11-25    23:46        5931744 gen_esp32part.exe
> -a---          2024-11-25    23:46        5923784 espota.exe
> -a---          2024-11-25    23:46           8119 platformio-build.py
> -a---          2024-11-25    23:46          15794 get.py
> -a---          2024-11-25    23:46           2260 gen_insights_package.py
> -a---          2024-11-25    23:46          24570 gen_esp32part.py
> -a---          2024-11-25    23:46           7824 gen_crt_bundle.py
> -a---          2024-11-25    23:46          10749 espota.py
> -a---          2024-11-25    23:46           4463 add_lib.sh
> 
> 
> # Instead of running the trojan maleware infected get.exe (by clicking), you can run the equivalent python code:
> python.exe .\get.py
> 
> # This will download and extract the follpowing:
> Downloading 'esp32-arduino-libs-c4ae7a78eafc339c71efcc7a75d5caa525c92f58.zip' ...
>     Renaming esp32-arduino-libs-c4ae7a78eafc339c71efcc7a75d5caa525c92f58 to esp32-arduino-libs ...
> Downloading 'xtensa-esp32-elf-12.2.0_20230208-x86_64-w64-mingw32.zip' ...
> Downloading 'xtensa-esp32s2-elf-12.2.0_20230208-x86_64-w64-mingw32.zip' ...
> Downloading 'xtensa-esp32s3-elf-12.2.0_20230208-x86_64-w64-mingw32.zip' ...
> Downloading 'xtensa-esp-elf-gdb-12.1_20231023-x86_64-w64-mingw32.zip' ...
> Downloading 'riscv32-esp-elf-12.2.0_20230208-x86_64-w64-mingw32.zip' ...
> Downloading 'riscv32-esp-elf-gdb-12.1_20231023-x86_64-w64-mingw32.zip' ...
> Downloading 'openocd-esp32-win32-0.12.0-esp32-20240318.zip' ...
> Downloading 'esptool-v4.6-win64.zip' ...
> Downloading 'x86_64-w64-mingw32.mklittlefs-c41e51a.200706.zip' ...
> Downloading 'mkspiffs-0.2.3-arduino-esp32-win32.zip'
> 
> # You will now have the following directories:
> 
> xtensa-esp32s3-elf
> xtensa-esp32s2-elf
> xtensa-esp32-elf
> xtensa-esp-elf-gdb
> riscv32-esp-elf-gdb
> riscv32-esp-elf
> pre-commit
> partitions
> openocd-esp32
> mkspiffs
> mklittlefs
> ide-debug
> esptool
> esp32-arduino-libs
> dist
> 
> # Close and Reopen Arduino-IDE
> ```

i tried this method but does not have the same file list that is shown to be extracted

jamesgbahr, 2024-12-04:
> The working method:
>
> ```shell
> #cd [YOUR_USER_NAME]/Documents/Arduino/hardware/heltec
> cd ~/Documents/Arduino/
>
> cd C:\Users\$env:USERNAME\Documents\Arduino
> mkdir hardware/heltec
> cd hardware/heltec [...]

i followed all these steps even downgrading to 1.1.15 and im still getting same error but for t190 vision master board 
"undefined reference to `ex_32k_start_cnt'"

eabase, 2025-08-08:
@RobinCutshaw 
Where's the "tools config"? 
Can you post the entire script, or link it?

RobinCutshaw, 2025-08-08:
In Arduino IDE, select Tools.

[image]