HelTecAutomation/Heltec_ESP32 #147: Cannot build with Lora Wifi 32 v2 unsing Platform.io

rodrigobutta, 2024-05-07:
This error happens in several lib files (not only in my main one)
**error: 'HELTEC_BOARD' was not declared in this scope**
And if I handpick fix it in heltec libraries (jsut for testing), some other similar ones ocurr too.

`platformio.ini`
```
[env:heltec_wifi_lora_32_V2]
platform = espressif32
board = heltec_wifi_lora_32_V2
framework = arduino
lib_deps = 
	heltecautomation/ESP32_LoRaWAN@^2.1.1
	heltecautomation/Heltec ESP32 Dev-Boards@^2.0.1
```

For what I see in this [Issue](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/248) it seems that the fix could be already merged a month ago, but I don´t see releases since a week before this fix ([Merge](https://github.com/HelTecAutomation/Heltec_ESP32/pull/133) on **March 27**, [last release](https://github.com/HelTecAutomation/Heltec_ESP32/tags) on **March 22** )

romoloman, 2024-05-07:
I'm really sorry for heltec, I tried before with a lot of satisfaction their cubecell series product, well documented and working flawlessy, I cannot really understand why they choose to customize the ESP32 IDF and not instead supplying a good library that relies on standard framework. To get the compiler be able to compile an example it took me three hours of work.
Even good hardware if badly documented becomes useless.
Installing the framework was a complete mess, i used a new virtual machine to do not destroy my setup.
The hardware is good but development environment is fool.

kainzjo1, 2024-05-28:
Same issue for me.

kainzjo1, 2024-05-28:
Compiling .pio\build\heltec_wifi_lora_32_V2\lib693\Heltec ESP32 Dev-Boards\driver\board.c.o
In file included from .pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/LoRaWan_APP.h:9,
                 from src/main.cpp:28:
.pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/loramac/LoRaMac.h:137: warning: "LORA_MAC_PRIVATE_SYNCWORD" redefined
 #define LORA_MAC_PRIVATE_SYNCWORD                   0x12

In file included from .pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/ESP32_Mcu.h:13,
                 from .pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/LoRaWan_APP.h:7,
                 from src/main.cpp:28:
.pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/driver/sx126x.h:122: note: this is the location of the previous definition
 #define LORA_MAC_PRIVATE_SYNCWORD                   0x1424

In file included from .pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/LoRaWan_APP.h:9,
                 from src/main.cpp:28:
.pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/loramac/LoRaMac.h:142: warning: "LORA_MAC_PUBLIC_SYNCWORD" redefined
 #define LORA_MAC_PUBLIC_SYNCWORD                    0x34

In file included from .pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/ESP32_Mcu.h:13,
                 from .pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/LoRaWan_APP.h:7,
                 from src/main.cpp:28:
.pio/libdeps/heltec_wifi_lora_32_V2/Heltec ESP32 Dev-Boards/src/driver/sx126x.h:127: note: this is the location of the previous definition
 #define LORA_MAC_PUBLIC_SYNCWORD                    0x3444

Compiling .pio\build\heltec_wifi_lora_32_V2\lib693\Heltec ESP32 Dev-Boards\driver\delay.c.o

AWeleczka, 2024-06-11:
I came across this https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/issues/248#issuecomment-2024632585 while looking for a solution to the same issue with the factory test example you provided above.

Downgrading to 1.1.5 allowed me to compile and upload the (different) factory test to my v2 board.
Though I'm using the Arduino IDE and not platformio I assume the library versions are the same.

felipemaion, 2024-06-19:
After digging a lot on this issue here is what I've found:
from ESP32_Mcu.h
```
/* HELTEC_BOARD
*
* ESP32S3 CHIP
* WIFI_LORA_32_V3          30
* WIRELESS_STICK_V3        31
* WIRELESS_STICK_LITE_V3   32
* WIRELESS_SHELL_V3        33
* WIRELESS_TRACKER         34
* WIFI_KIT_32_V3           35
*
* CAPSULE_SENSOR_V3 50
* WIRELESS_PAPER    60
*
*
* ESP32C3 CHIP
* WIRELESS_MINI_SHELL      70
*
*
* ESP32D0 CHIP
*
* WIFI_LORA_32_V2         0
* WIFI_LORA_32            1
* WIRELESS_STICK          2
* WIRELESS_STICK_LITE //WIRELESS_SHELL     3
* WIRELESS_BRIDGE         4
* WIFI_KIT_32             5
*/
```

And from "https://raw.githubusercontent.com/espressif/arduino-esp32/master/boards.txt":
``` 
heltec_wifi_lora_32_V2.build.defines=-D{build.band} -DMCU_ESP32_D0 -DWIFI_LORA_32_V2 -DHELTEC_BOARD=0 -DRADIO_CHIP_SX127X -DSLOW_CLK_TPYE=0 -DLoRaWAN_DEBUG_LEVEL={build.LoRaWanDebugLevel} -DACTIVE_REGION=LORAMAC_{build.band} -DLORAWAN_PREAMBLE_LENGTH={build.LORAWAN_PREAMBLE_LENGTH}  -DLORAWAN_DEVEUI_AUTO={build.LORAWAN_DEVEUI_AUTO}  {build.psram}
```

So, to make it work, my platformio.ini was like that:
```
[env:heltec_wifi_lora_32_V2]
platform = espressif32
lib_deps = heltecautomation/Heltec ESP32 Dev-Boards@^2.0.2
board = heltec_wifi_lora_32_V2
framework = arduino
upload_protocol = esptool
build_flags = 
	-D LoRaWAN_DEBUG_LEVEL=3
	-D LORAWAN_PREAMBLE_LENGTH=8
    -DMCU_ESP32_D0
    -D ARDUINO_HELTEC_WIFI_LORA_32_V2
	-D WIFI_LORA_32_V2
	-D HELTEC_WIFI_LORA_32_V2
	-D REGION_US915
	-D LORA_DISPLAY
    -DHELTEC_BOARD=0 
    -DRADIO_CHIP_SX127X 
    -DSLOW_CLK_TPYE=0

```

And the BUILD was successful. 
Cheers.

felipemaion, 2024-11-26:
> @felipemaion I don't have this file. Where can I find it? Or where should it be located if I don't have it?

Are you using Platformio on VSCODE to program your device? Usually the `platformio.ini` is on the root folder of the project.

On my machine it is located inside `<User>/Documents/PlatformIO/<ProjetectName>/platformio.ini` 
Hope this helps.

eabase, 2024-11-27:
@felipemaion 
Hmm, I'm  not using PlatformIO, I'm using Arduino-IDE... 
I thought platformio was just a filename... so maybe that explains it?

felipemaion, 2024-11-27:
> @felipemaion Hmm, I'm not using PlatformIO, I'm using Arduino-IDE... I thought platformio was just a filename... so maybe that explains it?

Yeap, that's explain it all. 
To run with Arduino-IDE it is much easier, you just need to install its library. 
Try to check this site: [Heltec Library](https://docs.heltec.org/en/node/esp32/esp32_general_docs/quick_start.html#installing-development-framework-and-library )

eabase, 2024-11-29:
@felipemaion 

> To run with Arduino-IDE it is much easier, you just need to install its library.

Yeah, I was using that for a while but with all the Heltec bugs in the examples, AIDE was getting painful to use, as the compilation for a simple LoRa example takes over 2 minutes to complete, while configuration is abstracted away in nightmare.  People say PlatformIO is a way faster, possibly 5x, and I like the simple clear configuration `*.ini` files. So I'm going to give that a shot.

* I have the `WiFi LoRa 32 (V3) v3.1`, what is the correct `Factory Test` example code to use?
* I also noticed that in `LoRa_rx_tx` example, that the device transmits all over the place, with bout 5x the spectral width of *correct* lora devices. What's going on?

thorsten-l, 2025-01-14:
```

thorsten-l, 2025-01-15:
https://github.com/thorsten-l/ESP32-HELTEC-LoRa-TTN-OTAA my working sample code

its-all-waves, 2026-09-22:
It's frustrating that info is not provided anywhere in the docs or in this repo.

Allows a build on the V4 board with the `LoRaWan` example:

```
build_flags =
	-D REGION_US915
	-D ACTIVE_REGION=LORAMAC_REGION_US915
	-D MCU_ESP32_S3
	-D HELTEC_BOARD=30
	-D SLOW_CLK_TPYE=0
	-D WIFI_LORA_32_V4
	-D RADIO_CHIP_SX1262
	-D LoRaWAN_DEBUG_LEVEL=0
	-D LORAWAN_PREAMBLE_LENGTH=8
	-D LORAWAN_DEVEUI_AUTO=0
	-D LORA_RX_LNA=1
	-D USE_NONE_PA
	-D HELTEC_WIFI_LORA_32_V4
	-D GPIO_PIN_COUNT=SOC_GPIO_PIN_COUNT
```