HelTecAutomation/Heltec_ESP32 #149: Wireless paper board issue with display and lorawan

arash-ahmadi, 2024-05-10:
Hello. I am trying the lorawan code on heltec wireless paper and it works fine. But when I add the code for e-ink display to it , both display and lorawan doesn't work. Please advise. Thanks

wangxiangwangeuse, 2024-05-20:
Using hspi and fspi

yrpn, 2024-07-15:
Is it possible to work together? Can you submit an example?

yrpn, 2024-07-15:
@wangxiangwangeuse Is it possible to work together? Can you submit an example?

Glenno-H, 2024-08-01:
Im having the same issues, using Heltec eink library 4.3.2 and ESP32 2.0.2

richardctrimble, 2025-09-30:
i don't know if this will help but i got the screen working on a esphome project i was working on ? https://github.com/[richardctrimble/esph-iboost-paper i finally uploaded it to GH