HelTecAutomation/Heltec_ESP32 #160: Infra Red library required for Heltec WiFi Lora 32 V3

mm5agm, 2024-08-08:
Hi all.
I'd like to move my stepper control program to Heltec WiFi Lora 32 V3 from my esp32.
In my esp32 program I use the Arduino official Infra red Library from https://github.com/Arduino-IRremote/Arduino-IRremote but it doesn’t work with ESP core greater than 2.0.17. The suggestion is to downgrade the ESP core to 2.0.17 but this stops the Heltec from working. Specifically  Mcu.begin(HELTEC_BOARD,SLOW_CLK_TPYE) fails as neither HELTEC_BOARD or SLOW_CLK_TPYE are declared. Does anyone know of a suitable Infra red library that works with esp core 3? I’m only looking to receive the IR code from my remote control.
Thanks
Colin

modernProteus, 2024-08-11:
You might try this:

#132 [https://github.com/HelTecAutomation/Heltec_ESP32/issues/132#issue-2209203528](url)

mm5agm, 2024-09-06:
IRremote by shirriff et al version 4.4.1 works. Download the zip if arduino only shows version 4.4.0
https://github.com/Arduino-IRremote/Arduino-IRremote

closed 2024-09-06 by mm5agm